Rahix/avr-hal #509: avrdude fails to process system wide configuration file

Tellurian-Ul, 2024-02-25:
I'm trying to perform the initial setup on my Windows 10 desktop using `winget` and then write the blink-led file to an old (5 years) Arduino UNO.

Everything was working fine until `avrdude` attempted to program to the board on COM3. I receive the error below:
```
avrdude error: unable to process system wide configuration file C:\Users\Jonath~1\AppData\Local\Temp\.avrdude-6P0nWT.conf
Error: avrdude failed
```

I tried reinstalling Rust and the other software packages to no avail. I also gave the vanilla Arduino IDE a try and that worked.

How should I troubleshoot this kind of issue?

stappersg, 2024-02-25:
> 
> How should I troubleshoot this kind of issue?
 
Start with asking questions like:
- What is the latest version?
- Am I using that version?
- What proofs that I am using the latest version?

Groeten
Geert Stappers
Did just wake up in UTC+1 and saw there is a new release,
does recommend `git pull` and `cargo build --release`.
-- 
Silence is hard to parse

stappersg, 2024-02-25:
On Sat, Feb 24, 2024 at 09:44:22PM -0800, Tellurian-Ul wrote:
> avrdude error: unable to process system wide configuration file
> C:\Users\Jonath~1\AppData\Local\Temp\.avrdude-6P0nWT.conf
> Error: avrdude failed
> 
> 
> How should I troubleshoot this kind of issue?
> 

Does the file exist?
Does the directory where it should be exist?

Groeten
Geert Stappers
-- 
Silence is hard to parse

Tellurian-Ul, 2024-02-25:
This is a fresh install, so I'm using the newest version of Ravedude (v0.1.7) and this repo.

> Does the file exist? Does the directory where it should be exist? Groeten Geert Stappers -- Silence is hard to parse

The folder mentioned does not exist, but it's in the temporary directory so it may have been deleted during the run. I don't know what should create this folder.

Tellurian-Ul, 2024-02-25:
Update:
After delving a bit, I found that invoking `avrdude` directly on the command line works just fine, but using `cargo run` (as the front page README suggest) always runs into this odd config file issue.

Using the command below allows me to flash after building:
```
avrdude -p m328p -c arduino -P COM3 -b 115200 -U flash:w:uno-test.elf
```
ref: https://www.youtube.com/watch?v=ZPSqhb4KKNc

Rahix, 2024-02-26:
ravedude is supplying its own configuration file and it seems your avrdude can't decipher it... What avrdude version do you have?

Tellurian-Ul, 2024-02-26:
> ravedude is supplying its own configuration file and it seems your avrdude can't decipher it... What avrdude version do you have?

ravedude v0.1.7 (no git)

stappersg, 2024-02-27:
On Mon, Feb 26, 2024 at 02:51:56PM -0800, Tellurian-Ul wrote:
> > ravedude is supplying its own configuration file and it seems your
> > avrdude can't decipher it... What avrdude version do you have?
> 
> ravedude v0.1.7 (no git)
> 

What avrdude version do you have?

-- 
Silence is hard to parse

tcordeu, 2024-02-28:
It also happened to me when I tried to use `ravedude@v0.1.7`, it works if you force `v0.1.6`:
```bash
 cargo +stable install ravedude@0.1.6
 ```
 @Tellurian-Ul

stappersg, 2024-03-03:
Here a happy Linux user.  As I understand the current status of this issue, does it now wait on effort by Microsoft Windows users.  The effort would be a _`git bisect` session_.

```text
stappers@paddy:/usr/src/rust/avr/avr-hal
$ git bisect start
status: waiting for both good and bad commits
stappers@paddy:/usr/src/rust/avr/avr-hal
$ git bisect bad
status: waiting for good commit(s), bad commit known
stappers@paddy:/usr/src/rust/avr/avr-hal
$ git bisect good ravedude-0.1.6
Bisecting: 27 revisions left to test after this (roughly 5 steps)
[ab3e4625fcbc0aa2dbdb1cb92596406f1da16cd8] remove `ufmt` re-export from `avr-hal-generic`
stappers@paddy:/usr/src/rust/avr/avr-hal
$ 
```

Rahix, 2024-03-03:
@stappersg, no.  The problem is that the new configuration for avrdude 7 seems to not work with avrdude 7.2.  I need to check what parts of the config are causing this and then check whether we can ship a single config for all avrdude 7.x versions or whether we need to get more specific and ship separate configs for each individual version...

stappersg, 2024-03-03:
> to check what parts of the config are causing this

Ravedude 1.6 does **not** write a config that breaks on avrdude 7.2 on Microsoft Windows.  Ravedude 1.7 does.  Those who never had a use case for`git bisect`, now have a great learning opportunity.

Rahix, 2024-03-03:
0.1.6 only works "by chance" because the avrdude 6 config it feeds into avrdude 7 seems to not entirely break things up to avrdude 7.2.  But upstream avrdude recommended that this is unsupported and we should be using a supported version of the configuration file.

There is no need for a bisection here because I can tell you exactly that commit 8137707b8783d2bee77f2cec4ff82e4dc0fd02e9 is the cause ;)

And by the way, this problem has nothing to do with Windows.  Confirmed the same behavior on Linux.

Rahix, 2024-03-12:
If someone who was affected by this issue can provide testing of #513, I can merge the change and cut a new release.

Rahix, 2024-03-15:
Fix is released as `ravedude` version 0.1.8.

PR #513 (2024-03-15): ravedude: Fix support for avrdude 7.0 to 7.2
Fixes #509.

@Tellurian-Ul, @tcordeu, please give this fix a spin.  You can install the patched version using

```bash
cargo install --git https://github.com/rahix/avr-hal --rev 67588b338b4f3abc370b7d05a45f70248d7f7818 ravedude
```