Rahix/avr-hal #515: Panic while trying to write to serial using `ufmt::uwriteln`

BiasedKiwi, 2024-03-07:
When I'm trying to write to serial using `ufmt::uwriteln` I get
```
thread '<unnamed>' panicked at C:\Users\Alexandre\.cargo\registry\src\index.crates.io-6f17d22bba15001f\ravedude-0.1.7\src\console.rs:35:32:
called `Result::unwrap()` on an `Err` value: Error { kind: InvalidData, message: "Windows stdio in console mode does not support writing non-UTF-8 byte sequences" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
Code:
```rs
#![no_std]
#![no_main]

use arduino_hal::prelude::*;
use panic_halt as _;

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);
    let mut serial = arduino_hal::default_serial!(dp, pins, 9600);

    ufmt::uwriteln!(&mut serial, "Hello, World!");

    loop {}
}
```
(follow up to #512)

Rahix, 2024-03-07:
Hi, there are two things happening here, I think:  First of all, ravedude is crashing with the error message you printed.  This is a bug and I've prepared #516 to hopefully fix it.  Please give this a try and report back what happens.

The second problem is that garbage data is received by ravedude in the first place.  This shouldn't be happening either.  I assume it is some problem with the serial connection, but let's wait on the output of the fixed ravedude from #516 - it will most likely make the issue easier to spot.

BiasedKiwi, 2024-03-07:
I switched to using #516 and this is what I got `?????????????????????????`

closed 2024-03-07 by Rahix

BiasedKiwi, 2024-03-07:
That was it! It does print `Hello, World!` now. Thanks for helping me out

PR #516 (2024-03-07): ravedude: Don't emit non-ascii bytes on Windows
Windows console stdio does not support sending UTF-8 sequences which means that we must ensure that any data received from the serial connection is cleaned to ASCII only before forwarding it to the console.

To make sure that users notice the illegal data getting sent, print `?` symbols instead.

Fixes #515.

@BiasedKiwi, please give this branch a try using

```bash
cargo install --git https://github.com/rahix/avr-hal --rev 3fddc711ad504c4dff4a18952834b8ad13d50446 ravedude
```