Rahix/avr-hal #526: Get pin number

wlwatkins, 2024-03-24:
Hi, 

I'm trying to port the firmata firmware in rust, more as an exercise to learn arduino coding than anything else. 

However, I required to know the pin number, indeed part of the protocol requires doing algebra with the pin number to send to the serial port e.g. `REPORT_ANALOG | (pin_number & 0xF)`.

However, when i instantiate a pin through `pin!`, the object does not give me access to the pin number.

I'm trying different ways, such as doing a match case to instantiate the pin in a custom struct to store the pin number, 

```rust
let pin = match pin_number {
            0 => pins.d0,
            1 => pins.d1,
            2 => pins.d2,
            ...
```

but i'm getting typing issues as there does not seem to be a generic pin type 

```bash
`match` arms have incompatible types
expected struct `avr_hal_generic::port::Pin<_, PD0>`
   found struct `avr_hal_generic::port::Pin<_, PD1>
```

is there a method, even `unsafe` to get back the pin number?

stappersg, 2024-03-24:
https://xyproblem.info/

stappersg, 2024-03-24:
On Sun, Mar 24, 2024 at 11:28:24AM -0700, Geert Stappers wrote:
> https://xyproblem.info/

Probably too implusive by me.

So now some elaboration:

I think^Wassume that original poster is trying
to shoehorn a typical C thing into Rust.
So the questions are about the shoehorning,
no about getting a Rust implementation of the firmata protocol.

wlwatkins, 2024-03-30:
For me here are X and Y
X = Learn how to do embedded programming in rust
Y = Port Firmata in rust

    - User wants to do X.
    - User doesn't know how to do X, but thinks they can fumble their way to a solution if they can just manage to do Y.
    - User doesn't know how to do Y either.
    - User asks for help with Y.
    - Others try to help user with Y, but are confused because Y seems like a strange problem to want to solve.
    - After much interaction and wasted time, it finally becomes clear that the user really wants help with X, and that Y wasn't even a suitable solution for X.

wlwatkins, 2024-03-31:
Hi, thanks for the comprehensive explanation. I was thinking about reimplementing some kid of struct to store each pin, which can later access the avr_hal API to work with it. But your approach is indeed interesting, and I appreciate the time you've put in the explanation. 
Regarding the modification of the Hal, my understanding is that is we want to shift the industry into using rust for embedded, wouldn't we need the same features to be available?
Anyway, I'll try this out thanks
thanks

Rahix, 2024-03-31:
> Regarding the modification of the Hal, my understanding is that is we want to shift the industry into using rust for embedded, wouldn't we need the same features to be available?

For the majority of usecases you don't need dynamic pin configuration which is why everyone is focusing on the static pin API.  It just allows catching a large number of potential bugs at compiletime which dynamic config cannot.

I agree that it would be nice to also provide a fully dynamic API, but it's just not as important for most users.

closed 2024-04-01 by wlwatkins