Rahix/avr-hal #535: cargo build - error[E0658] - proc_macro::Literal::byte_character(byte)

HaroPo, 2024-04-15:
Hi
I am a "baby beginner" with Rust and wanted to learn about Rust for embedded programming on 8-bit AVR MCUs...

I am facing errors at "cargo build" time and I wonder what I am supposed to do with the suggested 'help' lines.

In case you are interested to reproduce the issue that I am facing, I am providing the following information: 

After managing to get _RustUp_ installed and running, I started to deal with the following online resources information blogs to get thing setup:
- [https://www.youtube.com/watch?v=ZPSqhb4KKNc](url)
```
~$  sudo apt install -y  git  rust-all
~$  sudo apt install -y  libssl-dev  ; # Missing on my LinuxMint 21 system 
~$  mkdir -p /tmp/GitHub.com
~$  cd /tmp/GitHub.com/
/tmp/GitHub.com$  git clone  https://github.com/rust-lang/rustup.git 
/tmp/GitHub.com$  cd ./rustup/ 
/tmp/GitHub.com/rustup$  cargo build
/tmp/GitHub.com/rustup$  chmod +x ./rustup-init.sh
/tmp/GitHub.com/rustup$  ./rustup-init.sh
/tmp/GitHub.com/rustup$  sudo apt remove -y  rust-all  ; # Said better to get rid of LinuxMint Distro version...
/tmp/GitHub.com/rustup$  cargo build
/tmp/GitHub.com/rustup$  ./rustup-init.sh
/tmp/GitHub.com/rustup$  exit

~$  mkdir -p ~/Rust/projects/
~$  cd ~/Rust/projects/
~/Rust/projects$  cargo new
~/Rust/projects$  cd ./rust-arduino-blink/
~/Rust/projects/rust-arduino-blink$  
~/Rust/projects/rust-arduino-blink$  nano Cargo.toml
~/Rust/projects/rust-arduino-blink$  nano .cargo/config.toml
~/Rust/projects/rust-arduino-blink$  nano src/main.rs
~/Rust/projects/rust-arduino-blink$  rustup override set nightly-2021-01-07
~/Rust/projects/rust-arduino-blink$  cargo build
```
After a LOT a attempts, it always ended cycling infinitely with a bunch of errors whatever I changed as indicated. => Gave up + Go to bed .

- [https://medium.com/digitalfrontiers/rust-on-a-stm32-microcontroller-90fac16f6342](url)
```
~$  sudo apt install -y  gdb-multiarch 
~$  sudo apt install -y  libudev-dev  ; # Missing on my LinuxMint 21 system
~$  rustup update
~$  rustup component add llvm-tools-preview
~$  rustup target add thumbv7em-none-eabihf
~$  cargo install cargo-binutils cargo-embed cargo-flash cargo-expand
```
Having to try later with a STM32 Nucleo board taking time to come via mail...

- [https://medium.com/@ninjapiraatti/programming-arduino-with-rust-on-mac-7c7536f3973d](url)
```
~$  sudo apt install -y  avr-gcc  avrdude
~$  cargo +stable install ravedude
~$  cargo +stable install cargo-generate
~$  cargo generate --git https://github.com/Rahix/avr-hal-template.git
~$  cd ~/Rust/projects/rust-arduino-uno-blink/
~/Rust/projects/rust-arduino-uno-blink$  
~/Rust/projects/rust-arduino-uno-blink$  rustup toolchain list

    stable-x86_64-unknown-linux-gnu (default)
    nightly-2021-01-07-x86_64-unknown-linux-gnu
    nightly-2024-03-22-x86_64-unknown-linux-gnu (override)
    nightly-x86_64-unknown-linux-gnu

~/Rust/projects/rust-arduino-uno-blink$  cargo build
```
What ends into:
```
   Compiling rustc-std-workspace-core v1.99.0 (/home/hp-backups/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling proc-macro2 v1.0.80
   Compiling avr-hal-generic v0.1.0 (https://github.com/rahix/avr-hal?rev=21342dcace7184f01fdc4e9703b01197bd4b4b4f#21342dca)
   Compiling compiler_builtins v0.1.108
error[E0658]: use of unstable library feature 'proc_macro_byte_character'
   --> /home/hp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.80/src/wrapper.rs:871:21
    |
871 |                     proc_macro::Literal::byte_character(byte)
    |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #115268 <https://github.com/rust-lang/rust/issues/115268> for more information
    = help: add `#![feature(proc_macro_byte_character)]` to the crate attributes to enable
    = note: this compiler was built on 2024-03-21; consider upgrading it if it is out of date

error[E0658]: use of unstable library feature 'proc_macro_c_str_literals'
   --> /home/hp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.80/src/wrapper.rs:898:21
    |
898 |                     proc_macro::Literal::c_string(string)
    |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #119750 <https://github.com/rust-lang/rust/issues/119750> for more information
    = help: add `#![feature(proc_macro_c_str_literals)]` to the crate attributes to enable
    = note: this compiler was built on 2024-03-21; consider upgrading it if it is out of date

For more information about this error, try `rustc --explain E0658`.
error: could not compile `proc-macro2` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...

```
After searching clues in different ways, I ended with this:
- [https://www.google.com/search?q=add+`%23![feature(proc_macro_byte_character)]`+to+the+crate+attributes+to+enable](url)
- - [https://stackoverflow.com/questions/27454761/what-is-a-crate-attribute-and-where-do-i-add-it](url)

I still wonder where exactly to add those lines:
```
#![feature(proc_macro_byte_character)]
#![feature(proc_macro_c_str_literals)]
```
since the errors persist despite adding them to:
```
~/Rust/projects/rust-arduino-uno-blink/src/main.rs
```
or / and
```
~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.80/src/wrapper.rs
```
and also trying:
```
~/Rust/projects/rust-arduino-uno-blink$  cargo build -Z namespaced-features
```
... :thinking: 

Additionally, there is also some recent (8 hours ago) activity there:
- [https://github.com/rust-lang/rust/issues/115268](url)

I will then wait for news from the involved GitHub communities if unable to progress during the next coming hours.

Thank you for reading thus far.

Best regards,
  HP_

HaroPo, 2024-04-15:
I just re-ran RustUp Update, what didn't get listed in what I have done and nothing has changed since a 1~2 hours ago:
```
~/Rust/projects/rust-arduino-uno-blink$  rustup update

info: syncing channel updates for 'stable-x86_64-unknown-linux-gnu'
info: syncing channel updates for 'nightly-x86_64-unknown-linux-gnu'
info: checking for self-update

   stable-x86_64-unknown-linux-gnu unchanged - rustc 1.77.2 (25ef9e3d8 2024-04-09)
  nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.79.0-nightly (0d8b3346a 2024-04-14)

info: cleaning up downloads & tmp directories
```

HaroPo, 2024-04-15:
Adding the suggested 'help' lines:
```
#![feature(proc_macro_byte_character)]
#![feature(proc_macro_c_str_literals)]
```
to _/home/hp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/**proc-macro2-1.0.80/src/lib.rs**_, allowed to progress a little bit more, But still ended with errors:
```
~/Rust/projects/rust-arduino-uno-blink$  cargo build  -Z  build-std=core
   Compiling proc-macro2 v1.0.80
   Compiling quote v1.0.36
   Compiling syn v1.0.109
   Compiling avr-device-macros v0.5.4
   Compiling ufmt-macros v0.3.0
   Compiling avr-device v0.5.4
   Compiling ufmt v0.2.0
   Compiling avr-hal-generic v0.1.0 (https://github.com/rahix/avr-hal?rev=21342dcace7184f01fdc4e9703b01197bd4b4b4f#21342dca)
   Compiling atmega-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=21342dcace7184f01fdc4e9703b01197bd4b4b4f#21342dca)
   Compiling arduino-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=21342dcace7184f01fdc4e9703b01197bd4b4b4f#21342dca)
   Compiling rust-arduino-uno-blink v0.1.0 (/home/hp/Rust/projects/rust-arduino-uno-blink)
error[E0635]: unknown feature `proc_macro_byte_character`
 --> src/main.rs:4:12
  |
4 | #![feature(proc_macro_byte_character)]
  |            ^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0635]: unknown feature `proc_macro_c_str_literals`
 --> src/main.rs:5:12
  |
5 | #![feature(proc_macro_c_str_literals)]
  |            ^^^^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0635`.
error: could not compile `rust-arduino-uno-blink` (bin "rust-arduino-uno-blink") due to 2 previous errors

~/Rust/projects/rust-arduino-uno-blink$  
```

```
~/Rust/projects/rust-arduino-uno-blink$  rustc --explain E0635

The #![feature] attribute specified an unknown feature.

Erroneous code example:

#![feature(nonexistent_rust_feature)] // error: unknown feature
(END)

```

```
~/Rust/projects/rust-arduino-uno-blink$  cargo build  -Z  build-std=core
   Compiling rust-arduino-uno-blink v0.1.0 (/home/hp/Rust/projects/rust-arduino-uno-blink)
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
error: linking with `avr-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/hp/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/hp/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" VSLANG="1033" "avr-gcc" "-mmcu=atmega328p" "/tmp/rustcncTwfu/symbols.o" "/home/hp/Rust/projects/rust-arduino-uno-blink/target/avr-atmega328p/debug/deps/rust_arduino_uno_blink-a3946fdbd2fef60f.arduino_hal-1813dc6a7e8d8849.arduino_hal.e3a5162abf07a993-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "/home/hp/Rust/projects/rust-arduino-uno-blink/target/avr-atmega328p/debug/deps" "-L" "/home/hp/Rust/projects/rust-arduino-uno-blink/target/debug/deps" "-L" "/home/hp/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,-Bstatic" "/home/hp/Rust/projects/rust-arduino-uno-blink/target/avr-atmega328p/debug/deps/libcompiler_builtins-6d9348a04ce51620.rlib" "-Wl,-Bdynamic" "-lgcc" "-Wl,-z,noexecstack" "-L" "/home/hp/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-o" "/home/hp/Rust/projects/rust-arduino-uno-blink/target/avr-atmega328p/debug/deps/rust_arduino_uno_blink-a3946fdbd2fef60f.elf" "-Wl,--gc-sections"
  = note: /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: cannot find crtatmega328p.o: No such file or directory
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: cannot find -lm
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: cannot find -lc
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: cannot find -latmega328p
          collect2: error: ld returned 1 exit status
          

error: could not compile `rust-arduino-uno-blink` (bin "rust-arduino-uno-blink") due to 1 previous error

```
~/Rust/projects/rust-arduino-uno-blink$  
```

calebarapp, 2024-04-16:
I had this issue as well. I resolved it by updating the rust-toolchain.toml file to target v1.79.0-nightly like so: 

```
[toolchain]
channel = "nightly-2024-04-11"
components = [ "rust-src" ]
profile = "minimal"
```

HaroPo, 2024-04-16:
https://github.com/Rahix/avr-hal/issues/535#issuecomment-2058302706
> I had this issue as well. I resolved it by updating the rust-toolchain.toml file to target v1.79.0-nightly like so:
> 
> ```
> [toolchain]
> channel = "nightly-2024-04-11"
> components = [ "rust-src" ]
> profile = "minimal"
> ```

Thanks @calebarapp for the tip. :+1: 

This seems to risk to be cancelled out if you run '_**rustup update**_', isn't it ?

Maybe preferable to do the following:
```
~/path/to/actual/project$  rustup toolchain list  ; # To later compare for changes...
~/path/to/actual/project$  rustup override set nightly-yyyy-mm-dd
~/path/to/actual/project$  rustup toolchain list

    stable-x86_64-unknown-linux-gnu (default)
    nightly-2021-01-07-x86_64-unknown-linux-gnu
    nightly-2024-03-22-x86_64-unknown-linux-gnu (override)
    nightly-x86_64-unknown-linux-gnu

~/path/to/actual/project$
```
where you replace '_**yyyy-mm-dd**_'  by '_**2024-04-11**_' or whatever date you need.

Best regards,
HP_

Rahix, 2024-04-18:
There is a lot of chaotic information in this issue, not even sure where to start...

@HaroPo, it seems to me you have quite a few misunderstanding on how the entire thing is supposed to fit together.  We can clear these things up, but first let me focus on the issue at hand.

At the core stands a regression caused by `proc-macro2` which you unfortunately were one of the first to run into.  There are two possible solutions:

1. Upgrade to a later toolchain which we should probably try doing as soon as possible anyway.
2. Pin `proc-macro2` to an older version as to not have people's projects randomly breaking.

Rahix, 2024-04-18:
I've created a new issue #537 to track work on a solution.  This will keep things a bit cleaner than trying to do it here.

@HaroPo:  For your information, here are a few pointers to get you back on the right track:

> ```
> ~/Rust/projects$  cargo new
> ~/Rust/projects$  cd ./rust-arduino-blink/
> ```

Don't do this, use the [`avr-hal-template`](https://github.com/Rahix/avr-hal-template) instead.

> ```
> ~/Rust/projects/rust-arduino-blink$  rustup override set nightly-2021-01-07
> ```

That compiler version is entirely too old.  Stick to the one from the `avr-hal-template` which it defines here:

https://github.com/Rahix/avr-hal-template/blob/bb141bd00f6b9687e986db8aeb66a3f3f50cca12/rust-toolchain.toml#L2

You also should not set an override using rustup when a `rust-toolchain.toml` file is present.  rustup will automatically use the version from that file instead of the current default toolchain.

> ```
> ~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.80/src/wrapper.rs
> ```

Don't **ever** edit files in `.cargo/registry`, this is **never** the right way forward.  You will make things much worse this way.

> I just re-ran RustUp Update

`rustup update` has no effect when you have pinned a specific nightly using either `rust-toolchain.toml` or a `rustup override`.

>  The latest compilation error seems to be solved after I installed another missing software piece: 'avr-libc'

Yes, you need `avr-libc`.  Please check for all the other required dependencies in our README: https://github.com/Rahix/avr-hal?tab=readme-ov-file#quickstart

> Now, having to figure out about the 'cargo embed' "Error No supported probe was found" issue.

No, no, no, why are you trying to use `cargo embed`?  For Rust on AVR, you should use [`ravedude`](https://github.com/Rahix/avr-hal/tree/main/ravedude) our runner tool for AVR boards.  Again, this comes pre-configured in the `avr-hal-template` so all you need to do is `cargo run` and `ravedude` does the rest.

> This seems to risk to be cancelled out if you run rustup update, isn't it ?

No.  `rustup update` has no effect on this.

> Maybe preferable to do rustup override set nightly-yyyy-mm-dd

Avoid `rustup override`.  Instead, use a `rust-toolchain.toml` file which specifies what toolchain a project needs.

closed 2024-04-18 by Rahix

Rahix, 2024-05-02:
You can use the "discussions" in this project to ask user questions: https://github.com/Rahix/avr-hal/discussions

For general rust questions you can find help in the rust-lang forum: https://users.rust-lang.org/

And finally, there is an avr-rust matrix channel: https://matrix.to/#/#avr-rust_Lobby:gitter.im