Rahix/avr-hal #536: can't find crate for `core`

kacper-cholewinski, 2024-04-17:
I get this error when trying to compile:
error[E0463]:  can't find crate for `core`
  |
  = note: the `avr-atmega328p` target may not be installed
  = help: consider downloading the target with `rustup target add avr-atmega328p`

Working with freshly generated project, using cargo-generate and provided template.

Any help would be appreciated.

stappersg, 2024-04-17:
On Wed, Apr 17, 2024 at 11:41:33AM -0700, Kacper Cholewiński wrote:
> I get this error when trying to compile:
> error[E0463]:  can't find crate for `core`
>   |
>   = note: the `avr-atmega328p` target may not be installed
>   = help: consider downloading the target with [...]

`rustup target add avr-atmega328p`

> Any help would be appreciated.

Provide feedback

kacper-cholewinski, 2024-04-18:
Well, every avr target is custom made json, so it doesn't work with `rustup target add`

Rahix, 2024-04-18:
Hm, not finding core sounds like `build-std` is not active.  There should be a configuration file `.cargo/config.toml` with the following lines:

https://github.com/Rahix/avr-hal/blob/7f7d0a536d8045ed8f254fe75d9aa0d646c7a750/examples/arduino-uno/.cargo/config.toml#L9-L10

This tells cargo to build `core` for your specific target.

What version of `cargo` and `rustc` are you using?

stappersg, 2024-04-20:
> What version of `cargo` and `rustc` are you using?

And I would like to known additional information how to reproduce the error. Not for the _error_, but for _solving the error_.

kacper-cholewinski, 2024-04-20:
> What version of cargo and rustc are you using?
`cargo --version`: `cargo 1.77.2 (e52e36006 2024-03-26)`
`rustc --version`: `rustc 1.77.2 (25ef9e3d8 2024-04-09)`
`rustup --version`: 
```
rustup 1.27.0 (bbb9276d2 2024-03-08)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.79.0-nightly (0ad927c0c 2024-03-21)`
```
`rustup toolchain list`:
```
stable-x86_64-pc-windows-msvc
nightly-2023-08-08-x86_64-pc-windows-msvc
nightly-2024-03-22-x86_64-pc-windows-msvc (override)
nightly-x86_64-pc-windows-msvc (default)
```

There is `build-std = ["core"]` present in my config.toml.

> And I would like to known additional information how to reproduce the error.
There is not much to reproduce - I generated new project from template, for Arduino Uno, and used `cargo run` inside.

There is one thing that could be significant - I recently cleared my PATH env variable by accident, but I added **.../.cargo/bin** back, and it still doesn't work. Maybe there is some other path that I should add there again?

Rahix, 2024-04-20:
Hmm, can these numbers don't quite add up... Did you run all the `.. --version` commands inside the project directory?  What are the contents of your `rust-toolchain.toml`?

kacper-cholewinski, 2024-04-21:
> Did you run all the .. --version commands inside the project directory?

Yep.
> Hmm, can these numbers don't quite add up...

Well, it looks like rustc and cargo show latest stable release version, but below you can see that `rust-toolchain.toml` override works.
> What are the contents of your rust-toolchain.toml?

Here you go:

```
[toolchain]
channel = "nightly-2024-03-22"
components = ["rust-src"]
profile = "minimal"
```

Rahix, 2024-04-21:
I'd expect to see this:

```text
rustc 1.79.0-nightly (0ad927c0c 2024-03-21)
cargo 1.79.0-nightly (d438c80c4 2024-03-19)
```

Are you sure your `rustc`/`cargo` binaries are managed by `rustup`?  Does

```bash
rustc +nightly-2024-03-22 --version
```

work or do you get an error?

kacper-cholewinski, 2024-04-21:
Well, this is weird...
`rustc +nightly-2024-03-22 --version`: `rustc 1.77.2 (25ef9e3d8 2024-04-09)`

Rahix, 2024-04-21:
My best guess is that your rustup installation is broken in some way... Maybe delete all toolchains, reinstall rustup, and see if the problem persists.  Or ask rustup upstream about it...

kacper-cholewinski, 2024-04-21:
> Maybe delete all toolchains, reinstall rustup, and see if the problem persists.

Well, I did it, and everything works now.
Weird thing is, this was actually second time I reinstalled everything, after problem occured for the first time. Only difference is, I've chosen "complete" installation instead of "default" (in rustup installer for Windows). I also noticed that now rustup installed all of the components for given nightly version, not only "rust-src" like before. This way all my rustc, cargo, etc. versions are synchronized now.

Thanks for all your help, I appreciate it.

stappersg, 2024-04-21:
> Thanks for all your help, I appreciate it.

And thank you for the provided feedback.

What about closing this issue? If it should remain open, express for which reason.

Karma bonus points for _"comment and close"_  with a text like:

> On Microsoft windows version 9  installed Rust as advised for windows, `cargo install cargo-generate`, _more notes_, `cargo generate URL-to-template projectname`, `cd projectname` and `cargo run` worked for me on 2024-04-21.