Rahix/avr-hal #541: Very weird error when compiling a very basic program:

itamarsch, 2024-04-30:
[image]

```
#![no_std]
#![no_main]
use arduino_hal::clock::MHz16;
use arduino_hal::hal::delay::Delay;
use arduino_hal::{delay_ms, i2c};
use arduino_hal::{I2c, Peripherals};
use lcd_lcm1602_i2c::Lcd;
use numtoa::NumToA;
use panic_serial;
const PHI: f64 = 1.61803398;

type LCD<'a> = Lcd<'a, I2c, Delay<MHz16>>;
const LCD_ADDRESS: u8 = 0x3f;
//
panic_serial::impl_panic_handler!(
  arduino_hal::usart::Usart<
    arduino_hal::pac::USART0,
    arduino_hal::port::Pin<arduino_hal::port::mode::Input, arduino_hal::hal::port::PD0>,
    arduino_hal::port::Pin<arduino_hal::port::mode::Output, arduino_hal::hal::port::PD1>
  >
);

fn lcd_init<'a>(i2c: &'a mut I2c, delay: &'a mut Delay<MHz16>) -> Result<LCD<'a>, i2c::Error> {
    let mut lcd = lcd_lcm1602_i2c::Lcd::new(i2c, delay)
        .address(LCD_ADDRESS)
        .rows(2) // two rows
        .init()?;

    lcd.backlight(lcd_lcm1602_i2c::Backlight::On)?;
    lcd.clear()?;
    lcd.set_cursor(0, 0)?;
    Ok(lcd)
}

#[arduino_hal::entry]
fn main() -> ! {
    let dp = Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);
    let serial = arduino_hal::default_serial!(dp, pins, 57600);
    let serial = share_serial_port_with_panic(serial);

    let mut i2c = arduino_hal::I2c::new(
        dp.TWI,
        pins.a4.into_pull_up_input(),
        pins.a5.into_pull_up_input(),
        50000,
    );

    let mut delay = arduino_hal::Delay::new();

    let mut lcd = lcd_init(&mut i2c, &mut delay).unwrap();

    let mut i: f64 = 1.0;
    let mut buf = [0u8; 100];

    loop {
        i = libm::round(i * PHI);
        let u32v = i as i32;
        delay_ms(10);
        lcd.clear().unwrap();
        lcd.set_cursor(0, 0).unwrap();
        lcd.write_str(u32v.numtoa_str(10, &mut buf)).unwrap();
    }
}

```

stappersg, 2024-05-01:
I think:
- There should a git repository with the above `main.rs` **PLUS** `Cargo.toml` and such files for reproducablity
- This should be in a discussion

Patryk27, 2024-05-04:
It looks like I've accidentally forgotten to `#[avr_skip]` this intrinsic [way back in 2023](https://github.com/rust-lang/compiler-builtins/pull/527) 👀 

(tl;dr this function gets provided both by GCC's standard library and Rust's compiler-builtins, which creates a linking conflict)

I've created https://github.com/rust-lang/compiler-builtins/pull/601 to fix this; after that gets merged, I'll bump compiler-builtins within rustc and ping back, as always 🙂

Patryk27, 2024-05-07:
[Right here](https://github.com/rust-lang/rust/blob/cc8d9b692720c7d15445024485c4bbfce35dbce6/library/std/Cargo.toml#L20) :-)

itamarsch, 2024-05-11:
> It looks like I've accidentally forgotten to `#[avr_skip]` this intrinsic [way back in 2023](https://github.com/rust-lang/compiler-builtins/pull/558) 👀 
> 
> 
> 
> (tl;dr this function gets provided both by GCC's standard library and Rust's compiler-builtins, which creates a linking conflict)
> 
> 
> 
> I've created https://github.com/rust-lang/compiler-builtins/pull/601 to fix this; after it gets merged, I'll bump compiler-builtins within rustc and ping back, as always 🙂 

Have bumped up compiler-builtins?

Patryk27, 2024-05-12:
[I'm on it](https://github.com/rust-lang/rust/pull/124886), but there are some (AVR-unrelated) crashes that I need to investigate.

Patryk27, 2024-05-18:
Status: together with my AVR fix, bumping compiler-builtins brought some extra (non-AVR-related) f16 & f128 changes which need special handling in the compiler, so it's not as easy of a merge as usual - the current pull request is:

https://github.com/rust-lang/rust/pull/125016

Patryk27, 2024-08-08:
Status: same as https://github.com/Rahix/avr-hal/issues/566#issuecomment-2276373883.

Patryk27, 2024-09-24:
Okie, it should work on the newest toolchain 🙂