Rahix/avr-hal #542: Ravedude freezes on programmer uploading

rhaskia, 2024-05-04:
I'm trying to upload the template file onto an Arduino nano with the command `ravedude nano -cb 57600 target\avr-atmega328p\release\metrics.elf -P COM4`. Unfortunately, the command stops during the programming stage and doesn't seem to continue, despite the elf file only being about 1kb and the Arduino ide working fine for uploads.

Rahix, 2024-05-04:
Please post the full output from ravedude, otherwise we have no clues as to what might be going on here...

rhaskia, 2024-05-04:
The output is just 
```Board Arduino Nano 
Programming <file>=> COM5
```
I managed to finally get something flashed onto the nano, but only through first converting it to a hex and then uploading it using the following commands:
```
avr-objcopy -j .text -j .data -O ihex target\avr-atmega328p\debug\metrics.elf sketch.ino.hex
avrdude -p atmega328p -F -P COM5 -c arduino -U flash:w:sketch.ino.hex
```
Does ravedude convert to hex first or have I set up something wrong somewhere?

Rahix, 2024-05-04:
So you are not seeeing _any_ avrdude output after the `Programming ...` line?

Please provide the following info:

- `ravedude --version`
- `avrdude -?`
- Operating System

rhaskia, 2024-05-04:
Yes that's right.
ravedude v0.1.8 (no git)
```
Usage: avrdude [options]
Options:
  -p <partno>            Specify AVR device; -p ? lists all known parts
  -p <wildcard>/<flags>  Run developer options for matched AVR devices,
                         e.g., -p ATmega328P/s or /S for part definition
  -b <baudrate>          Override RS-232 baud rate
  -B <bitclock>          Specify bit clock period (us)
  -C <config-file>       Specify location of configuration file
  -C +<config-file>      Specify additional config file, can be repeated
  -N                     Do not load config file
  -c <programmer>        Specify programmer; -c ? and -c ?type list all
  -c <wildcard>/<flags>  Run developer options for matched programmers,
                         e.g., -c 'ur*'/s for programmer info/definition
  -A                     Disable trailing-0xff removal for file/AVR read
  -D                     Disable auto erase for flash memory; implies -A
  -i <delay>             ISP Clock Delay [in microseconds]
  -P <port>              Connection; -P ?s or -P ?sa lists serial ones
  -r                     Reconnect to -P port after "touching" it; wait
                         400 ms for each -r; needed for some USB boards
  -F                     Override invalid signature or initial checks
  -e                     Perform a chip erase
  -O                     Perform RC oscillator calibration (see AVR053)
  -t                     Run an interactive terminal when it is its turn
  -T <terminal cmd line> Run terminal line when it is its turn
  -U <memstr>:r|w|v:<filename>[:format]
                         Carry out memory operation when it is its turn
                         Multiple -t, -T and -U options can be specified
  -n                     Do not write to the device whilst processing -U
  -V                     Do not automatically verify during -U
  -E <exitsp>[,<exitsp>] List programmer exit specifications
  -x <extended_param>    Pass <extended_param> to programmer, see -xhelp
  -v                     Verbose output; -v -v for more
  -q                     Quell progress output; -q -q for less
  -l logfile             Use logfile rather than stderr for diagnostics
  -?                     Display this usage

avrdude version 7.3, https://github.com/avrdudes/avrdude
```
Windows

Rahix, 2024-05-05:
Can you add `--debug-avrdude` to the `ravedude` invocation in `.cargo/config.toml` please?  That should print out the entire avrdude command that ravedude tries to call.  You can then try calling that avrdude command manually and maybe play with it to see if anything about it is wrong.

> Does ravedude convert to hex first

Ravedude doesn't convert to hex because you can also pass an ELF to avrdude directly if you add `:e` after the filename.

stappersg, 2025-01-01:
Request: Add label `MSwindows`
Proposal: Close this issue, there was six months silence

rhaskia, 2025-01-01:
Oh sorry completely forgot about this issue, pretty sure I solved it somehow as I did manage to get it uploaded.

closed 2025-01-01 by rhaskia

Rahix, 2025-01-04:
> pretty sure I solved it somehow as I did manage to get it uploaded.

In the future, it would be nice to comment what your solution was.  Others are taking their (free!) time to help you here, so be kind and give back by documenting the solution for the next person who runs into this.

rhaskia, 2025-01-05:
I can't remember exactly but I believe cutting down the program's size was all I did. I am not sure why I left the issue all of a sudden though, sorry for that.