Rahix/avr-hal #547: Cant Build WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.

BrandonLeeDotDev, 2024-05-09:
On ubuntu 23, GCC 13

Can push from Arduino IDE

```
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
```

```
nightly-2024-03-22-x86_64-unknown-linux-gnu (directory override for '/home/user/Documents/project_transmit/avr-hal/examples/arduino-nano')
```

Rahix, 2024-05-09:
Please provide more information.  What's not working?  What behavior are you observing?  What did you do to trigger this?

You can't expect us to guess what is going on from this little information.

BrandonLeeDotDev, 2024-05-09:
@Rahix updated...

stappersg, 2024-05-10:
> updated...

Please make a better attempt.

stappersg, 2024-05-10:
On Thu, May 09, 2024 at 11:18:19PM -0700, Geert Stappers wrote:
> > updated...
> 
> Please make a better attempt.
> 

And while you are at it, retitle to
  avrdude error: programmer is not responding

Groeten
Geert Stappers
-- 
Silence is hard to parse

Rahix, 2024-05-10:
Also, please check whether you have the issue discussed in #299.

BrandonLeeDotDev, 2024-05-10:
@Rahix 

<pre><font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/target/avr-atmega328p/debug</b></font>$ avrdude --version
...
avrdude version 7.1, URL: &lt;https://github.com/avrdudes/avrdude&gt;
</pre>

<pre><font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/target/avr-atmega328p/debug</b></font>$ ravedude --version
ravedude v0.1.8 (no git)
</pre>

Cargo.toml
```
[build]
target = "../../avr-specs/avr-atmega328p.json"

[target.'cfg(target_arch = "avr")']
runner = "ravedude new-nano -cb 115200 --debug-avrdude"

[unstable]
build-std = ["core"]
```

after adding debug flag above && new-nano. I had already figured out the baud rate was incorrect... 
<pre><font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/examples/arduino-nano</b></font>$ cargo build
<font color="#00FF00"><b>   Compiling</b></font> compiler_builtins v0.1.108
<font color="#00FF00"><b>   Compiling</b></font> core v0.0.0 (/home/user/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
<font color="#00FF00"><b>   Compiling</b></font> proc-macro2 v1.0.79
<font color="#00FF00"><b>   Compiling</b></font> unicode-ident v1.0.12
<font color="#00FF00"><b>   Compiling</b></font> syn v1.0.109
<font color="#00FF00"><b>   Compiling</b></font> rustversion v1.0.16
<font color="#00FF00"><b>   Compiling</b></font> embedded-hal-bus v0.1.0
<font color="#00FF00"><b>   Compiling</b></font> paste v1.0.15
<font color="#00FF00"><b>   Compiling</b></font> quote v1.0.36
<font color="#00FF00"><b>   Compiling</b></font> avr-hal-generic v0.1.0 (/home/user/Documents/project_transmit/avr-hal/avr-hal-generic)
<font color="#00FF00"><b>   Compiling</b></font> ufmt-macros v0.3.0
<font color="#00FF00"><b>   Compiling</b></font> avr-device-macros v0.5.4
<font color="#00FF00"><b>   Compiling</b></font> rustc-std-workspace-core v1.99.0 (/home/user/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
<font color="#00FF00"><b>   Compiling</b></font> nb v1.1.0
<font color="#00FF00"><b>   Compiling</b></font> vcell v0.1.3
<font color="#00FF00"><b>   Compiling</b></font> critical-section v1.1.2
<font color="#00FF00"><b>   Compiling</b></font> bare-metal v1.0.0
<font color="#00FF00"><b>   Compiling</b></font> cfg-if v1.0.0
<font color="#00FF00"><b>   Compiling</b></font> void v1.0.2
<font color="#00FF00"><b>   Compiling</b></font> ufmt-write v0.1.0
<font color="#00FF00"><b>   Compiling</b></font> embedded-hal v1.0.0
<font color="#00FF00"><b>   Compiling</b></font> embedded-storage v0.2.0
<font color="#00FF00"><b>   Compiling</b></font> unwrap-infallible v0.1.5
<font color="#00FF00"><b>   Compiling</b></font> cfg-if v0.1.10
<font color="#00FF00"><b>   Compiling</b></font> avr-device v0.5.4
<font color="#00FF00"><b>   Compiling</b></font> ufmt v0.2.0
<font color="#00FF00"><b>   Compiling</b></font> panic-halt v0.2.0
<font color="#00FF00"><b>   Compiling</b></font> nb v0.1.3
<font color="#00FF00"><b>   Compiling</b></font> embedded-hal v0.2.7
<font color="#00FF00"><b>   Compiling</b></font> atmega-hal v0.1.0 (/home/user/Documents/project_transmit/avr-hal/mcu/atmega-hal)
<font color="#00FF00"><b>   Compiling</b></font> arduino-hal v0.1.0 (/home/user/Documents/project_transmit/avr-hal/arduino-hal)
<font color="#00FF00"><b>   Compiling</b></font> arduino-nano-examples v0.0.0 (/home/user/Documents/project_transmit/avr-hal/examples/arduino-nano)
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
<font color="#00FF00"><b>    Finished</b></font> `dev` profile [optimized + debuginfo] target(s) in 9.13s
</pre>

Something to note here is it is compiling however... there is not change when I upload to the board...
<pre><font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/target/avr-atmega328p/debug</b></font>$ avrdude -p m328p -c arduino -P /dev/ttyUSB0 -b 115200 flash:w:nano-blink.elf

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude done.  Thank you.</pre>

Rahix, 2024-05-10:
Sorry, I'm not following.  What is the problem at this point?  You show `cargo build`.  Are you saying that `cargo run` (i.e. running ravedude) still produces the same error as above with the `nano-new` config?

BrandonLeeDotDev, 2024-05-10:
@Rahix before the initial post it was compiling. Then it ceased. Until that point I assumed fault. After, I filed the issue. That said here is the output when trying to upload to the board from the elf

<pre><font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/examples/arduino-nano</b></font>$ cargo clean
<font color="#00FF00"><b>     Removed</b></font> 410 files, 205.4MiB total
<font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/examples/arduino-nano</b></font>$ cargo build
<font color="#00FF00"><b>   Compiling</b></font> compiler_builtins v0.1.108
<font color="#00FF00"><b>   Compiling</b></font> core v0.0.0 (/home/user/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
<font color="#00FF00"><b>   Compiling</b></font> proc-macro2 v1.0.79
<font color="#00FF00"><b>   Compiling</b></font> unicode-ident v1.0.12
<font color="#00FF00"><b>   Compiling</b></font> syn v1.0.109
<font color="#00FF00"><b>   Compiling</b></font> rustversion v1.0.16
<font color="#00FF00"><b>   Compiling</b></font> embedded-hal-bus v0.1.0
<font color="#00FF00"><b>   Compiling</b></font> paste v1.0.15
<font color="#00FF00"><b>   Compiling</b></font> quote v1.0.36
<font color="#00FF00"><b>   Compiling</b></font> avr-hal-generic v0.1.0 (/home/user/Documents/project_transmit/avr-hal/avr-hal-generic)
<font color="#00FF00"><b>   Compiling</b></font> ufmt-macros v0.3.0
<font color="#00FF00"><b>   Compiling</b></font> avr-device-macros v0.5.4
<font color="#00FF00"><b>   Compiling</b></font> rustc-std-workspace-core v1.99.0 (/home/user/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
<font color="#00FF00"><b>   Compiling</b></font> nb v1.1.0
<font color="#00FF00"><b>   Compiling</b></font> critical-section v1.1.2
<font color="#00FF00"><b>   Compiling</b></font> void v1.0.2
<font color="#00FF00"><b>   Compiling</b></font> embedded-hal v1.0.0
<font color="#00FF00"><b>   Compiling</b></font> bare-metal v1.0.0
<font color="#00FF00"><b>   Compiling</b></font> vcell v0.1.3
<font color="#00FF00"><b>   Compiling</b></font> cfg-if v1.0.0
<font color="#00FF00"><b>   Compiling</b></font> ufmt-write v0.1.0
<font color="#00FF00"><b>   Compiling</b></font> unwrap-infallible v0.1.5
<font color="#00FF00"><b>   Compiling</b></font> embedded-storage v0.2.0
<font color="#00FF00"><b>   Compiling</b></font> cfg-if v0.1.10
<font color="#00FF00"><b>   Compiling</b></font> ufmt v0.2.0
<font color="#00FF00"><b>   Compiling</b></font> panic-halt v0.2.0
<font color="#00FF00"><b>   Compiling</b></font> avr-device v0.5.4
<font color="#00FF00"><b>   Compiling</b></font> nb v0.1.3
<font color="#00FF00"><b>   Compiling</b></font> embedded-hal v0.2.7
<font color="#00FF00"><b>   Compiling</b></font> atmega-hal v0.1.0 (/home/user/Documents/project_transmit/avr-hal/mcu/atmega-hal)
<font color="#00FF00"><b>   Compiling</b></font> arduino-hal v0.1.0 (/home/user/Documents/project_transmit/avr-hal/arduino-hal)
<font color="#00FF00"><b>   Compiling</b></font> arduino-nano-examples v0.0.0 (/home/user/Documents/project_transmit/avr-hal/examples/arduino-nano)
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
<font color="#00FF00"><b>    Finished</b></font> `dev` profile [optimized + debuginfo] target(s) in 9.49s
<font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/examples/arduino-nano</b></font>$ cargo run --bin nano-blink
WARN rustc_codegen_ssa::back::link Linker does not support -no-pie command line option. Retrying without.
<font color="#00FF00"><b>    Finished</b></font> `dev` profile [optimized + debuginfo] target(s) in 0.04s
<font color="#00FF00"><b>     Running</b></font> `ravedude new-nano -cb 115200 --debug-avrdude /home/user/Documents/project_transmit/avr-hal/target/avr-atmega328p/debug/nano-blink.elf`
thread &apos;main&apos; panicked at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ravedude-0.1.8/src/main.rs:97:47:
board not found
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
<font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/target/debug</b></font>$ cd ../../avr-atmega328p/debug
<font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/target/avr-atmega328p/debug</b></font>$ ls
<font color="#0CD819"><b>build</b></font>  <font color="#0CD819"><b>examples</b></font>     nano-adc.d    nano-blink.d    nano-panic.d
<font color="#0CD819"><b>deps</b></font>   <font color="#0CD819"><b>incremental</b></font>  <font color="#00FF00"><b>nano-adc.elf</b></font>  <font color="#00FF00"><b>nano-blink.elf</b></font>  <font color="#00FF00"><b>nano-panic.elf</b></font>
<font color="#00FF00"><b>user@terminal</b></font>:<font color="#0CD819"><b>~/Documents/project_transmit/avr-hal/target/avr-atmega328p/debug</b></font>$ avrdude -p m328p -c arduino -P /dev/ttyUSB0 -b 115200 flash:w:nano-blink.elf

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude done.  Thank you.
</pre>

BrandonLeeDotDev, 2024-05-10:
@Rahix Adding the debug flag, nano-new, baud rate... etc changed nothing.. there is actually no additional info created using the debug flag as shown above... or am I just missing something?

Rahix, 2024-05-10:
You need to actually read the error messages you are getting!

> ```
> thread 'main' panicked at [...]/ravedude-0.1.8/src/main.rs:97:47:
> board not found
> ```

And we can see the reason in the line above:  You wrote `new-nano` instead of `nano-new`.  Please try again with the correct name ;)

BrandonLeeDotDev, 2024-05-10:
@Rahix That was it! Thank you for the help! :)

closed 2024-05-10 by BrandonLeeDotDev