Rahix/avr-hal #554: Cannot find `pwm` in `embedded_hal`

Sciancisco, 2024-05-29:
I have a problem importing `SetDutyCycle`.

Here are my imports:
```
use arduino_hal::prelude::*;
use arduino_hal::simple_pwm::*;
use embedded_hal::pwm::SetDutyCycle;
use embedded_hal::serial::{Read, Write};
```

Here is my error:
```
error[E0432]: unresolved import `embedded_hal::pwm`
 --> src/main.rs:9:19
  |
9 | use embedded_hal::pwm::SetDutyCycle;
  |                   ^^^ could not find `pwm` in `embedded_hal`
```

I built the documentation and *I* can find `embedded_hal::pwm`. I looked at the source of `embedded_hal` linked in the doc and it does declare `pub mod pwm`. `embedded_hal` is in the dependencies of my `Cargo.toml`. Therefore, rustc should find the pwm module.

But it gets weirder (to me at least) because looking at this same source code, it shouldn't be able to find the `serial` module because it is not declared in `embedded_hal`. In fact, search through all the doc, there is no module `serial` at all !..

I've managed to trace back the `Read` and `Write` traits to `arduino_hal::prelude::{_embedded_hal_serial_Read, _embedded_hal_serial_Write}`, but their source code link is broken as it tries to get `embedded_hal/serial.rs.html`, the non existent `embedded_hal::serial`...

And what's with `embedded_hal` and `embedded_hal_v0`? Because in `avr_hal_generic` it imports the `Read` and `Write` traits from v0 but v0 doesn't appear to exist anyway, not even in the `Cargo.lock`...

I am stumped that it finds a module that it shouldn't and  cannot find a module that it should :laughing: :thinking:

Rahix, 2024-05-30:
You are probably confused because we interface with two versions of `embedded-hal`:

- `embedded_hal` is the new v1.0 version that was released a few months back
- `embedded_hal_v0` is the older v0.# version that was used by everyone before that. we still support it because there are many drivers that still target the older version only.

The relevant dependency definitions are here:

https://github.com/Rahix/avr-hal/blob/38549cc4142881d302374535b3621578ffccaff2/avr-hal-generic/Cargo.toml#L26-L33

You probably need to import the older version of `embedded-hal` to access the traits that you need.

Sciancisco, 2024-05-31:
It worked! Thank you! I didn't know this cargo trick.

And thank you a lot for this library! First timer in embedded and with its documentation and examples, it is relatively easy to use.

closed 2024-05-31 by Sciancisco