Rahix/avr-hal #555: peripherals type alias

djdisodo, 2024-05-30:
i think peripherals type names are hard to write and remember currently,

it makes some problem because rust requires you to put type on generic parameters, function arguments or static/const values

for example how can tell what type of pin d9 is? `atmega_hal::port::PB1`

however names should be different depending on the board

first, maybe we can introduce some macro that parses given value and return the type

iirc this is used in syn with `token!` macro

or we can just create type alias

Rahix, 2024-06-08:
> for example how can tell what's type of pin d9 on UNO is? atmega_hal::port::PB1

This is a good point.  I think the `avr_hal_generic::renamed_pins! {}` macro should generate type aliases for the pins at least.  This also shouldn't be too difficult to achieve.

Rahix, 2024-06-08:
(Created #558 to track this)

PR #559 (2024-06-10): arduino-hal: provide renamed types matching fields
A very simple improvement to fix #558 and fix #555.

The approach is a basic application of renaming with `paste!`, but I choose to generate slightly different code than suggested:

```rust
pub struct Pins {
    // Notice the lack of documentation here, I thought it became redundant...
    pub a0: PinA0<
        mode::Input<mode::Floating>,
    >,
    // ...
}

// ...since I moved it to the type aliases themselves.

/// `A0`
///
/// * ADC0 (ADC input channel 0)
/// * PCINT8 (pin change interrupt 8)
pub type A0 = atmega_hal::port::PC0;
// ...

 [... cut]
commit afdadebd (2024-06-10): arduino-hal: provide renamed types matching fields

A very simple improvement to fix #558 and fix #555.
closed 2024-06-10 by Rahix