Rahix/avr-hal #566: Issue with Floating Point math to serial print out

TimKotowski, 2024-07-14:
`use panic_halt as ;
use arduino_hal::I2c;
use bme280_rs::{Bme280, Configuration, Oversampling, SensorMode};
use fixed::types::U16F16;
use micromath::F32Ext;
use ufmt::{uDisplay, uWrite};

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);
    let mut serial = arduino_hal::default_serial!(dp, pins, 57600);
    let mut d2 = arduino_hal::Delay::new();

    let mut d = arduino_hal::Delay::new();
    let i2c = I2c::new(
        dp.TWI,
        pins.a4.into_pull_up_input(),
        pins.a5.into_pull_up_input(),
        50000,
    );
    let mut bme280 = Bme280::new_with_address(i2c, 0x77, d);

    bme280.init().unwrap();
    bme280.set_sampling_configuration(
        Configuration::default()
          .with_temperature_oversampling(Oversampling::Oversample1)
          .with_pressure_oversampling(Oversampling::Oversample1)
          .with_humidity_oversampling(Oversampling::Oversample1)
          .with_sensor_mode(SensorMode::Normal),
    ).unwrap();

    loop {
        let pre = bme280.read_sample().unwrap().pressure;
        if let Some(pressure) = pre {
            let ratio = pressure / 1013.25;
            let exponent = 1.0 / 5.255;
            let a0 = 44330.0 * (1.0 - ratio.powf(exponent));
            let f = ufmt_float::uFmt_f32::Zero(a0);

            ufmt::uwriteln!(&mut serial, "Pressure fixed: {}", f).unwrap()
        } else {
            ufmt::uwriteln!(&mut serial, "NNOE").unwrap();
        }

        arduino_hal::delay_ms(100);
    }
}`

seems to be getting error 
`LC_ALL="C" PATH="/Users/tim/.rustup/toolchains/nightly-2024-03-22-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/bin:/Users/tim/.cargo/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Users/tim/.cargo/bin:/usr/local/go/bin" VSLANG="1033" "avr-gcc" "-mmcu=atmega328p" "/var/folders/m0/kb5xzxn5223dmc_ksg7pzlvw0000gn/T/rustczLzpwo/symbols.o" "/Users/tim/bme_test/target/avr-atmega328p/debug/deps/bme_test-f3c0e98f60d647e8.ufmt_float-1131307c75409628.ufmt_float.6b526f773fefe65f-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "/Users/tim/bme_test/target/avr-atmega328p/debug/deps" "-L" "/Users/tim/bme_test/target/debug/deps" "-L" "/Users/tim/.rustup/toolchains/nightly-2024-03-22-aarch64-apple-darwin/lib/rustlib/avr-atmega328p/lib" "-Wl,-Bstatic" "/Users/tim/bme_test/target/avr-atmega328p/debug/deps/libcompiler_builtins-922b355f64723ac9.rlib" "-Wl,-Bdynamic" "-lgcc" "-Wl,-z,noexecstack" "-L" "/Users/tim/.rustup/toolchains/nightly-2024-03-22-aarch64-apple-darwin/lib/rustlib/avr-atmega328p/lib" "-o" "/Users/tim/bme_test/target/avr-atmega328p/debug/deps/bme_test-f3c0e98f60d647e8.elf" "-Wl,--gc-sections" "-no-pie"
  = note: /opt/homebrew/opt/avr-binutils/bin/avr-ld: /opt/homebrew/Cellar/avr-gcc@9/9.4.0_1/lib/avr-gcc/9/gcc/avr/9.4.0/../../../../../../avr/lib/avr5/libm.a(addsf3.o): in function `__addsf3':
          (.text.avr-libc.fplib+0x2): multiple definition of `__addsf3'; /Users/tim/bme_test/target/avr-atmega328p/debug/deps/libcompiler_builtins-922b355f64723ac9.rlib(compiler_builtins-922b355f64723ac9.compiler_builtins.a4206a2074977fb4-cgu.1.rcgu.o):/Users/tim/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.108/src/macros.rs:500: first defined here
          collect2: error: ld returned 1 exit status`

I did see floats when being print out are an issue but with `ufmt_float` should solve the issue, but when i include the line `            ufmt::uwriteln!(&mut serial, "Pressure fixed: {}", f).unwrap()` this is where i get an error. I previously added `# This is needed because of a known bug https://github.com/Rahix/avr-hal/issues/125#issuecomment-771578390
[profile.dev.package.compiler_builtins]
overflow-checks = false` but this did not solve it. Unaware of this issue else where.

Rahix, 2024-07-16:
CC @Patryk27

TimKotowski, 2024-07-16:
update, adding `rustflags = ["-C", "link-arg=-Wl,--allow-multiple-definition"]` allows the error to go away, 

but as far as calculations are concerned, seems off 
```
let ratio: f32 = (986.82 ) / 1013.25f32;
            let exponent: f32 = 1.0 / 5.255f32;
            let a0: f32 = 44330.0 * (1.0 - ratio.powf(exponent));
            println!("bitchin outloud {}", a0 as u32);
```
should be 222 but i get 426. Unrelated to avr_hal as this point, I am using micromath crate, guess just weird precision with the ardunio microcontroller?

Patryk27, 2024-07-21:
I think this is going to be similar to https://github.com/Rahix/avr-hal/issues/541, which is stuck on https://github.com/rust-lang/rust/pull/125016 (tl;dr I've added a small fix to compiler_builtins, but then currently we can't bump compiler_builtins within rustc, because it brings f16 & f128, which have missing support within the compiler...).

Anyway, I'm watching the compiler_builtins bump and once it's done, I'll revisit this issue (unless someone has a better lead, but that's my hunch for now).

tgross35, 2024-07-21:
> I think this is going to be similar to #541, which is stuck on [rust-lang/rust#125016](https://github.com/rust-lang/rust/pull/125016) (tl;dr I've added a small fix to compiler_builtins, but we can't bump compiler_builtins within rustc at the moment, because it brings f16 & f128 as well, and those have missing support within the compiler...).

FYI f16 & f128 will be disabled in compiler_builtins by default, with an exception on platforms that are known to work (currently x86 and aarch64).

I have a fix for the compiler builtins issue up and waiting for review, hopefully soonish.

Patryk27, 2024-08-08:
Status: seems to compile on the latest toolchain, but due to https://github.com/Rahix/avr-hal/issues/571 (aka https://github.com/llvm/llvm-project/issues/102436) the entire newest toolchain is unusable for AVR anyway, so the issue's stalled on that.

Patryk27, 2024-09-24:
Okie, it should work on the newest toolchain 🙂