Rahix/avr-hal #571: Serial writes aren't working on `rustc 1.82.0-nightly (8e86c9567 2024-08-01)`

hankertrix, 2024-08-02:
After updating the rustc compiler to `rustc 1.82.0-nightly (8e86c9567 2024-08-01)` from `rustc 1.82.0-nightly (c1a6199e9 2024-07-24)`, serial writes don't work properly any more.

Here is the minimal reproducible example for the Arduino Mega 2560:
```rust
#![no_std]
#![no_main]

use panic_halt as _;

#[arduino_hal::entry]
fn main() -> ! {
    let peripherals = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(peripherals);

    let mut serial = arduino_hal::default_serial!(peripherals, pins, 57600);
    ufmt::uwrite!(serial, "Serial initialised!\n").unwrap();
    loop {}
```

The console prints out `Serial initialised!` with a new line at the back as expected on `rustc 1.82.0-nightly (c1a6199e9 2024-07-24)`, but only prints `S` on `rustc 1.82.0-nightly (8e86c9567 2024-08-01)`. Nothing is printed to the console if other things are being initialised before the serial handler, such as a timer or a stepper motor driver.

I believe this should also be reproducible on the Arduino Uno as well.

CoolSlimbo, 2024-08-02:
By comparing a diff of the generated assembly, you can see the difference.
```diff
- cpi r25, 0
- breq .LBB1_1
+ cpi r25, 1
+ brne .LBB1_1
```

The prior rust version (commit `8e86c9567`) changes it to a "branch if not equal" op. I am by no means an assembly expert (first time ever looking at AVR instruction set), but, as it seems, this is saying to branch is the operation before (the `cpi r25, 1`) is not true.

Odds are, I've very much misinterpreted that, however. So, in place of the fact I have no clue what I'm doing, I'm going to call the one person who I believe can help. @Patryk27. This seems like a you kind of thing, if you don't mind.

Patryk27, 2024-08-02:
Thanks for pinging, I'll take a look later!

Patryk27, 2024-08-02:
Alright, so that comparison is actually correct - it's like `if r25 == 0 { ... }` vs `if !(r25 == 1) { ... }`, which do the same thing (given that `r25 ∈ {0, 1}`).

This codegen difference stems from some changes within rustc / llvm optimizations, because it used to generate:
```llvm
  %_9.not = icmp eq i8 %1, 0
  br i1 %_9.not, label %bb11, label %bb10
```
... but since 2024-08-01 it generates this instead:
```llvm
  %_9 = trunc nuw i8 %1 to i1
  br i1 %_9, label %bb10, label %bb11
```

I think both snippets are correct and what caught my attention were the jumps - all `brne` etc. in the invalid binary are offset by two bytes, which looks very wrong. I'll try to narrow the issue down.

Patryk27, 2024-08-07:
Status: I've confirmed it's a regression with LLVM (building newest rustc locally with LLVM 18 yields a correct binary), will try to `git bisect` it.

Patryk27, 2024-08-08:
Alright, it looks like the breaking commit was https://github.com/llvm/llvm-project/commit/6859685a87ad093d60c8bed60b116143c0a684c7 (or, more precisely, https://github.com/llvm/llvm-project/commit/84428dafc0941e3a31303fa1b286835ab2b8e234, because the former has a small compilation issue fixed by this commit).

Now, onto discovering why an unrelated change broke AVR 🔍

Patryk27, 2024-08-08:
It seems that the good version hits this branch:

https://github.com/llvm/llvm-project/blob/876cbf9a127b0a18374dcf91c969c7ba152e048c/llvm/lib/Target/AVR/MCTargetDesc/AVRMCCodeEmitter.cpp#L98

... while on the invalid commit, the flow goes right here:

https://github.com/llvm/llvm-project/blob/876cbf9a127b0a18374dcf91c969c7ba152e048c/llvm/lib/Target/AVR/MCTargetDesc/AVRMCCodeEmitter.cpp#L108

Looking further 🔍

Patryk27, 2024-08-08:
Got a lead, the difference between both versions boils down to this check:

https://github.com/llvm/llvm-project/blob/1919db907bc3e0541839992d903b6b875da98b04/llvm/lib/Target/AVR/MCTargetDesc/AVRAsmBackend.cpp#L521

This function returns `true` for the working version, but `false` nowadays - and, actually, judging by what it's doing, it was supposed to return `false` before as well, it was only now that this check became correct 🔍

Patryk27, 2024-08-19:
Status: patch is ready (https://github.com/llvm/llvm-project/pull/102936), waiting for review.

Tellurian-Ul, 2024-09-01:
Unrelated to the fix. I'm working on a project that's encountering this bug. 

How do I rebuild using the old version of `rustc` nightly?

hankertrix, 2024-09-01:
@Tellurian-Ul, navigate to your project root folder and run the command below:

```shell
rustup override set nightly-2024-07-25
```

Patryk27, 2024-09-12:
Status: on my way backporting the fix to rust's fork of llvm-project.

Patryk27, 2024-09-24:
Alright, fixed in the newest toolchain 🙂