Rahix/avr-hal #577: Can't specify different upload and monitor serial ports.

paulwrath1223, 2024-09-05:
The Sparkfun Pro Micro (and presumably the other 32u4 boards) use a different COM port depending on if it's in bootloader mode. 
I'm currently trying to just get a blink sketch working on a Pro Micro, but cant get it working. I am totally liable to be making a giant blunder, but I can only upload by specifying the COM port, and then the program immediately crashes after uploading with an unspecified error.

Tested on Arduino Uno and without changing anything it works flawlessly.

stappersg, 2024-09-05:
On Wed, Sep 04, 2024 at 10:55:15PM -0700, Paul Fornage wrote:
> 
> I am totally liable to be making a giant blunder

Please continue, you will^Wmight learn a lot

 
> Tested on Arduino Uno and without changing anything it works flawlessly.

According https://store.arduino.cc/en-nl/products/arduino-uno-rev3
has an Arduino Uno an ATmega328p. And that is not a 32u4.

Groeten
Geert Stappers
-- 
Silence is hard to parse

paulwrath1223, 2024-09-05:
> According https://store.arduino.cc/en-nl/products/arduino-uno-rev3 has an Arduino Uno an ATmega328p. And that is not a 32u4. 

I should have been more specific, I have 2 boards.

When creating a project for the Uno, everything works fine (except the 'issue' described here: https://github.com/Rahix/avr-hal/issues/248)

However, for the Pro Micro, I need to specify a COM port for it to upload successfully, and the serial output port is different.

Rahix, 2024-09-14:
Yeah, I'm aware of this shortcoming in the current design.  I agree that it should be possible to specify an alternative port for the console and essentially disconnect the console logic from the flashing logic a lot more.

Let's take care of this after #522 is merged.

closed 2025-09-27 by Rahix