Rahix/avr-hal #578: Ravedude Console Timeout

hcabel, 2024-09-08:
Hey newbie here, I'm trying to use the console but it timeout 2s after it launched, with `"The device does not recognize the command. (os error 22)"`

# Context:
I'm using **Window 11**
with a **promicro Atmega32u4** USB-C
I use this value: **`runner = "ravedude -cb 57600 promicro -P COM8"`**

I've tweak the code a little bit to get rid of the timeout and I've added a print, (that's how I got the error message) but none of that helped. 😞 

# My guess
My only guess (pure theory) maybe it's because my controller is getting disconnected.
When I press the reset switch that's when my COM8 port is showing up. But then when the new firmware is installed it goes away _(code 45, hardware not connected)._
My Controller is still connected though, but maybe it use another port or something.

#### **I'll really appreciate any help** 😸 🙏

Rahix, 2024-09-14:
Hi,

you've run into a problem that isn't quite obvious:  The ATmega32U4-based (e.g. Leonardo, Promicro) boards work differently from the ATmega328P-based ones (e.g. Uno, Mega).  The ATmega32U4 handles USB directly instead of using a simple UART for serial communication which then gets translated to USB by an external component.  This means that you have to do USB communication on the ATmega32U4-based boards to get a serial console.  On other boards this is much easier because UART is much simpler than USB.

Unfortunately, USB is not yet fully supported here in avr-hal.  See issue #40...

As an alternative, you can connect an external USB-to-UART bridge to the UART pins (RX,TX) of your board and use that one for serial communication.

Dehnert86, 2025-01-17:
Hi,

Try to reduce the Baud rate. maybe. 9600