Rahix/avr-hal #582: Compiler Rejects Floats (Defined twice?)

Tellurian-Ul, 2024-09-17:
Possibly related to #566.

I'm unable to compile when floats are introduced to my code. I'm currently relying on `nightly-2024-07-25` given the broken state of the compiler. I've posted the error I'm seeing upon `cargo build` below. My best guess is that the nightly override somehow created two definitions of floating point operations?

```
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-mmcu=atmega328p" "C:\\User\\AppData\\Local\\Temp\\rustcZfgPO9\\symbols.o" "C:\\User\\project\\target\\avr-atmega328p\\debug\\deps\\p2_temp_record-0e3cb22c69115045.dht11-23173e3f7edbd950.dht11.3e7d5a656d0319e8-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "C:\\User\\project\\target\\avr-atmega328p\\debug\\deps" "-L" "C:\\User\\project\\target\\debug\\deps" "-L" "C:\\User\\.rustup\\toolchains\\nightly-2024-07-25-x86_64-pc-windows-msvc\\lib\\rustlib\\avr-atmega328p\\lib" "-Wl,-Bstatic" "C:\\Userr\\project\\target\\avr-atmega328p\\debug\\deps\\libcompiler_builtins-d2fcaf6386bb018a.rlib" "-Wl,-Bdynamic" "-lgcc" "-Wl,-z,noexecstack" "-L" "C:\\User\\.rustup\\toolchains\\nightly-2024-07-25-x86_64-pc-windows-msvc\\lib\\rustlib\\avr-atmega328p\\lib" "-o" "C:\\User\\project\\target\\avr-atmega328p\\debug\\deps\\p2_temp_record-0e3cb22c69115045.elf" "-Wl,--gc-sections" "-no-pie"
  = note: C:/User/AppData/Local/Microsoft/WinGet/Packages/ZakKemble.avr-gcc_Microsoft.Winget.Source_8wekyb3d8bbwe/avr-gcc-13.2.0-x64-windows/bin/../lib/gcc/avr/13.2.0/../../../../avr/bin/ld.exe: C:/User/AppData/Local/Microsoft/WinGet/Packages/ZakKemble.avr-gcc_Microsoft.Winget.Source_8wekyb3d8bbwe/avr-gcc-13.2.0-x64-windows/bin/../lib/gcc/avr/13.2.0/../../../../avr/lib/avr5\libm.a(addsf3.o): in function `__addsf3':
          (.text.avr-libc.fplib+0x2): multiple definition of `__addsf3'; C:\User\project\target\avr-atmega328p\debug\deps\libcompiler_builtins-d2fcaf6386bb018a.rlib(compiler_builtins-d2fcaf6386bb018a.compiler_builtins.fee158677b5537c-cgu.13.rcgu.o):C:\User\.cargo\registry\src\index.crates.io-6f17d22bba15001f\compiler_builtins-0.1.109/src/macros.rs:500: first defined here
          collect2.exe: error: ld returned 1 exit status

error: could not compile `project` (bin "project") due to 1 previous error
```

Does anyone know if there's a fix that allows floats?

Rahix, 2024-09-20:
Is this a duplicate of #541 maybe?

Tellurian-Ul, 2024-09-21:
> Is this a duplicate of #541 maybe?

Looks like it. Probably safe to close this issue.

stappersg, 2024-09-22:
> Probably safe to close this issue.

Please do.

@Tellurian-Ul  do known that on _github.com_  one can close the issues that are opened by themself.

closed 2024-09-22 by Rahix