Rahix/avr-hal #584: Conformity of SPI implementation with embedded_hal::spi::SpiBus trait

dmholtz, 2024-09-24:
Structs implementing the `SpiBus` trait from `embedded_hal`  should be unaware of the chip select (CS) pin in order to allow bus sharing and implement additional traits such as `SpiDevice`, see [embedded_hal docs](https://docs.rs/embedded-hal/latest/embedded_hal/spi/index.html).

Thus, I believe that the current design of the [`Spi` struct](https://github.com/Rahix/avr-hal/blob/544e50bdf2c1944281d9f63fa9ef3ccf68b5f684/avr-hal-generic/src/spi.rs#L193) does not provide a appropriate base for implementing th `SpiBus` trait, as it is tightly coupled to the type of the CS pin.

Rahix, 2024-09-24:
I think you misunderstood why our `Spi` struct deals with a CS pin.  It does not actually make use of the CS pin in any way - in fact the `Spi::new` constructor passes back the CS pin to the user right away.  This odd behavior is needed because we must enforce that the CS pin is always an output pin - if it isn't, the hardware SPI peripheral does not behave correctly.

Whether you later use the wrapped CS pin as an actual CS or for different purposes is up to you.  You can also use any other pin as a CS, or none at all.  This matches the expectations of `embedded-hal`.

This hardware restriction was originally discussed in #27.

K4LCIFER, 2025-09-03:
Is the following error related to this issue?

```
error[E0277]: the trait bound `avr_hal_generic::spi::Spi<Atmega, SPI, PB5, PB3, PB4, PB2>: embedded_hal::spi::SpiDevice` is not satisfied
   --> master/src/main.rs:24:39
    |
 24 |     let mut nrf24 = NRF24L01::new(ce, spi).unwrap();
    |                     -------------     ^^^ the trait `embedded_hal::spi::SpiDevice` is not implemented for `avr_hal_generic::spi::Spi<Atmega, SPI, PB5, PB3, PB4, PB2>`
    |                     |
    |                     required by a bound introduced by this call
    |
```
I'm trying to get [this crate](https://github.com/astro/embedded-nrf24l01) working, but I'm having issues trying to get atmega-hal's SPI device to work with it.