Rahix/avr-hal #593: `winget` makes first build a lot more complicated

Pumbakos, 2024-10-20:
I just started using Windows after 2 year break and I am new to rust ecosystem, I was using macOS (where I could use `brew` for those who don't know) so I started using `winget` as documentation states. 
I've encountered an issue where `avr-gcc` could not interpret `--print-memory-usage` flag. At first glance, I did not suspect versioning because "Hey, I just installed it". 
The error was `error: linking with ‘avr-gcc’`, so I started poking around using some `rustflags` and various `linkers` because that's what some of the stackoverflows answers have said. Then I realized that `winget` installed `winavr-20100110` that seems to be quite old. I followed [this great article by Tony Mitchell](https://www.tonymitchell.ca/posts/setup-avr-toolchain-on-windows/#install-avr-toolchain) and turned out to be working. 

This is the line that caused all that [(link to README.md)](https://github.com/Rahix/avr-hal/blob/15080aa41e726bc720adf36ce0fc5a030390c822/README.md?plain=1#L24):
```shell
winget install AVRDudes.AVRDUDE ZakKemble.avr-gcc
```

stappersg, 2024-10-20:
On Sun, Oct 20, 2024 at 12:37:11PM -0700, Dawid Pałubiak wrote:
> using Windows 
> This is the line that caused all that
> ```shell
> winget install AVRDudes.AVRDUDE ZakKemble.avr-gcc
> ```

patches welcome.

Slightly longer version:

Acknowlegde on "It does not work on my platform".
Do not expect that "works on **all platforms**" is a top priority.

Please update yourself the README.md with how to install winget.
(A link to "How to install winget on Microsoft Windows 11" will do.)

Groeten
Geert Stappers
-- 
Silence is hard to parse

Rahix, 2024-10-21:
Agreed.  I don't have a Windows system to check with so I rely on feedback from users on what works.  If the current instructions have problems, please send a PR to update/fix them :)

2bndy5, 2025-01-01:
[`winget` is rather tied to the Windows store](https://github.com/microsoft/winget-cli#installing-the-client). Early builds of Windows 10 didn't have it yet. Latest Win 10 & 11 should have it installed if it has Windows Store installed.

If using an enterprise edition of Windows (with no Windows store or `winget` installed), you may have to seek alternate methods. By default, you can't use `winget` in CI with github's `windows-latest` runner ([attempted](https://github.com/Cyberboss/install-winget/blob/1c6f189175e9f015bf1aa113cd1cc6c73efb9d47/action.yml#L22)).

---
Not sure if this helps
```
winget search ZakKemble.avr-gcc --versions
Found AVR-GCC [ZakKemble.avr-gcc]
Version
-------
14.1.0
13.2.0
12.1.0
```
```
winget search AVRDudes.AVRDUDE --versions
Found AVRDUDE [AVRDudes.AVRDUDE]
Version
-------
7.3
7.2
7.1
```