Rahix/avr-hal #597: Moteino Board Support

powturns, 2024-11-02:
I've been experimenting with adding support for the [moteino](https://lowpowerlab.com/guide/moteino/).
This board uses an ATmega328p chip, but has the unique feature of exposing ADC7/ADC6 as pins A7/A6.

I'm wondering how I can define these pins using `avr_hal_generic::renamed_pins!`

Board definitions: https://lowpowerlab.github.io/MoteinoCore/Moteino1.6.4.zip
Platform.io https://docs.platformio.org/en/stable/boards/atmelavr/moteino.html

[image]
(from https://lowpowerlab.com/guide/moteino/out-of-box/)

[image]
(from https://lowpowerlab.com/guide/moteino/design-files/)

[image]
(from https://github.com/LowPowerLab/Moteino/blob/master/R6/)

Rahix, 2024-11-02:
(Transferred to `avr-hal` as this issue is not related to `avr-device`)

You cannot access A6/A7 via the usual pins struct.  It works differently, see here for example:

https://github.com/Rahix/avr-hal/blob/15080aa41e726bc720adf36ce0fc5a030390c822/examples/nano168/src/bin/nano168-adc.rs#L46-L51

powturns, 2024-11-02:
Thanks very much!

closed 2024-11-02 by powturns