Rahix/avr-hal #600: cargo run - Error: Failed reading avrdude version information.

alter-kaker, 2024-11-10:
Trying to set up a Rust toolchain for Arduino because I'm no good at all with C/C++.

I'm using the `cargo generate --git https://github.com/Rahix/avr-hal-template.git` template. Build goes fine, but cargo-run gives the following output:
```
$ cargo run
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.17s
     Running `ravedude nano -cb 57600 target/avr-atmega328p/debug/mediaplayer.elf`
Error: Failed reading avrdude version information.

Caused by: No such file or directory (os error 2)```

I'm running Arch (EndeavorOS), fully updated as of today.

Thanks!

alter-kaker, 2024-11-10:
I needed to install avrdude.

closed 2024-11-10 by alter-kaker

Rahix, 2024-11-11:
I guess we could add a check that avrdude is even there before trying to read out the version.  The error message would then be a lot less confusing!