Rahix/avr-hal #601: ravedude: Add a check if avrdude is even installed

Rahix, 2024-11-11:
Add a check to ravedude on whether avrdude is even available.  If not, the error message should give pointers on where to get the correct avrdude version.

(From #600)

AlternativleyCoolSlimbo, 2025-03-25:
I'll claim this.
And this time, I won't vanish off the face of the earth, leaving someone else to pick up the slack (sorry Rahix, and thank you armandas)!

To add to your initial part of the request, are you suggesting different error messages, dependant on platform, similar to what's already present in the readme?

Rahix, 2025-03-26:
> To add to your initial part of the request, are you suggesting different error messages, dependant on platform, similar to what's already present in the readme?

Yeah, something that helps the particular user on getting their setup working.  So platform-specific for sure and ideally all the necessary steps to get it running.  For Linux, I'd just cover the distros we mention already. Any special-distro users should be advanced enough to figure it out themselves.

AlternativleyCoolSlimbo, 2025-03-26:
Roger roger. I'll also include making sure `avr-gcc` is mentioned for the error message.

closed 2025-04-13 by Rahix