Rahix/avr-hal #622: i2cdetect in the read direction hangs/crashes after finding a sensor

sp4ghet, 2025-01-01:
I'm using an Arduino Mega 2560 R3 with a MH-Z16 CO2 sensor in I2C mode

I basically just took the sample for I2Cdetect verbatim (just swapped the order of Read/Write)

```rust
#![no_std]
#![no_main]

use arduino_hal::{prelude::*, I2c};
use panic_halt as _;

const NDIR_ADDR: u8 = 0x4D;

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);

    let mut serial = arduino_hal::default_serial!(dp, pins, 57600);

    let mut led = pins.d13.into_output();
    let mut i2c = arduino_hal::I2c::new(
        dp.TWI,
        pins.d20.into_pull_up_input(),
        pins.d21.into_pull_up_input(),
        100000,
    );

    ufmt::uwriteln!(&mut serial, "Read direction test:\r").unwrap_infallible();
    i2c.i2cdetect(&mut serial, arduino_hal::i2c::Direction::Read)
        .unwrap_infallible();

    ufmt::uwriteln!(&mut serial, "Write direction test:\r").unwrap_infallible();
    i2c.i2cdetect(&mut serial, arduino_hal::i2c::Direction::Write)
        .unwrap_infallible();

    loop{};
}
```

```
/// serial output
Read direction test:
-    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:       -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- 4d
```

It seems to run until it finds the sensor, and then it freezes.
the Write direction seems to run fine, but the read direction hangs.
I've tried adding some code below the i2cdetect but it doesn't appear as if any of it actually runs.

I've tried using the normal Arduino/C++ code provided by the manufacturer and it worked (connected and sent measurements over serial) so I don't think I've setup the hardware incorrectly, but please let me know if I fumbled something obvious.

I also tried just calling `ping` and it seemed to behave the same way (unsurprisingly..) where the ping goes through, but all following calls to i2c hang/panic(?)

```rust
    let pinged = i2c
         .ping_device(NDIR_ADDR, arduino_hal::i2c::Direction::Read)
         .unwrap();
    ufmt::uwriteln!(&mut serial, "Pinged: {}\r", if pinged { 1 } else { 0 }).unwrap();

    ufmt::uwriteln!(&mut serial, "Read direction test:\r").unwrap_infallible();
// this code doesn't run (except the first row of labels)
    i2c.i2cdetect(&mut serial, arduino_hal::i2c::Direction::Read)
        .unwrap_infallible();
```

sp4ghet, 2025-01-01:
If it is necessary this is the product page (links to datasheets and sample code)

https://github.com/SandboxElectronics/NDIR/tree/master/NDIR_I2C

https://sandboxelectronics.com/?product=mh-z16-ndir-co2-sensor-with-i2cuart-5v3-3v-interface-for-arduinoraspeberry-pi

stappersg, 2025-01-01:
My advice to sp4ghet:
* Close this issue
* `cargo generate --git https://github.com/Rahix/avr-hal-template.git --name ndir_mvp`
* Keep the [mvp](https://en.wikipedia.org/wiki/Minimum_viable_product) a _minimum viable product_, meaning: Stay away from _swapping the order of Read/Write_ and other _let me change this now, I will need it later away_
* Check, double check and triple check the hardware setup. That implies having a schematic of the electronics. And above all:  Understand all the hardware involved
* Document your prototype further. Do it for your future self and do it for your current self that is seeking help from others.
* Enjoy the challenge of getting to work what you envision
* Persist
* Smile humbly upon reaching your goal
* Maybe publish the achievement  (It is 2025, there is something called _Internet_, every one is a publisher.)

My advice to everyone seeking help with **their** project:
* Understand the gist of http://www.catb.org/~esr/faqs/smart-questions.html
* Aim for win win

bdew, 2025-05-26:
I've ran into this too, it looks like something in the i2c implementation breaks if you request a read but don't actually read anything.

Adding a read stops it from getting stuck.

Rahix, 2025-06-05:
> Adding a read stops it from getting stuck.

The intention is to _not_ do this because doing actual reads/writes can have side-effects for the remote devices.  Ideally, we just want to adress the device, check whether it responds and then abort the transaction.  If this is broken right now, we have to investigate why...

Rahix, 2025-09-25:
> > But I wonder about the impl of the new [I2c::transaction](https://docs.rs/embedded-hal/latest/embedded_hal/i2c/trait.I2c.html#tymethod.transaction). The contract defined upstream does not forbid empty data buffers. So we could potentially hit the same illegal situation here ([L452](https://github.com/Rahix/avr-hal/pull/681/files#diff-ad5e049947ba3f5c9b2e51324221c391961c6db2df7809ea9140704895e4da4aL452)). I do not think the forced stop is really appropriate, unless this is really the behavior embedded-hal wants here. But silently not doing the whole transaction because it is empty does not look better either. What are your thoughts?
> 
> Even if the contract does not require full non-empty data buffers, it still requires that the transaction is performed or to return an error. As there is no proper/meaningful way for us to handle the transaction in this case, we should return an error. We should also check upfront that all the buffers passed to I2c::transaction() are not empty, and return an error before transmitting anything (avoid cancelling in the middle of the transaction), i guess?

I think there is more nuance here: When you chain multiple reads, it is quite fine when there is one in there with a zero-sized buffer.  Allowing that can make downstream-code more robust and easy to write for dynamic transfer lengths.

I think a good starting point is to also check what other HAL crates are doing with this.  I poked the upstream issue in e-h about the topic as well so let's see what they have to say.

---

In any case, I'll merge #665 now to resolve the immediate issue in `ping_device()` for starters.