Rahix/avr-hal #632: ravedude nix flake is out of date.

tuxerator, 2025-01-24:
When trying to build `ravedude` with nix I get the following error.
```
error: package `toml v0.8.11` cannot be built because it requires rustc 1.70 or newer, while the currently active rustc version is 1.67.1
Either upgrade to rustc 1.70 or newer, or use
cargo update -p toml@0.8.11 --precise ver
where `ver` is the latest version of `toml` supporting rustc 1.67.1
[naersk] cargo returned with exit code 101, exiting
```

This is fixable by updating the `flake.lock` file but it might be better to point naersk to the `rust-toolchain.toml` file.

Rahix, 2025-01-25:
Hi,

feel free to send a PR.  I can't maintain the nix support, so I have to rely on others for keeping it up-to-date :)

I agree that it would be best if the version from `rust-toolchain.toml` is used automatically.

LuigiPiucco, 2025-05-02:
I can take care of this as a Nix user, but before I'd like to ask: could we additionally have a development shell at the root, to make getting the compiler easier? Especially since we need somewhat specific nightlies, this could be very helpful. If yes, I'll send it as an additional commit in the PR. It would consist of an additional `flake.nix` and `flake.lock` at the root.

Rahix, 2025-05-03:
> could we additionally have a development shell at the root, to make getting the compiler easier?

Sure, no objection from me :)

PR #655 (2025-05-03): Nix update and revamp
Fix #632. I asked there if the second commit of this PR would be accepted, but let's move the discussion here.

This PR updates the ravedude flake and revises it to follow `ravedude/rust-toolchain.toml`. It also adds the latest stable date to the toolchain, to avoid silently using the older version if it's not in the locked `fenix` input (also because it is generally good to specify the date).

Beyond that, it adds a Nix development shell to the root, which similarly follows `rust-toolchain.toml` and provides runtime programs like avr-gcc too. It does not build the projec [... cut]
closed 2025-05-03 by Rahix