Rahix/avr-hal #635: avrdude ignoring parameters passed in

ErickTorresBrownU, 2025-02-27:
Running
`ravedude nano --port COM4 --baudrate 115200 --open-console --debug-avrdude`
we get
`Dbg.Command avrdude -c arduino -p atmega328p -P COM4 -b 57600 -e -D -U flash:w:target\avr-atmega328p\debug\arduino-rust-playground.elf:e`

showing that avrdude ignores the explicit baudrate argument passed in. This is an issue as it prevents correct flashing.

A current workaround is to just replace "nano" with "uno" and proceed, but this shouldn't be necessary.

Rahix, 2025-02-27:
`--baudrate` is just the serial console baudrate, not the baudrate for flashing.  There are two different versions of the nano which require different baudrates.  `nano` is the config for the 57600 variant, `nano-new` is the config for the 115200 variant.  So you should use the `nano-new` for your board.

ErickTorresBrownU, 2025-02-27:
Thanks! Appreciate the clarification

closed 2025-02-27 by ErickTorresBrownU

stappersg, 2025-02-27:
On Thu, Feb 27, 2025 at 11:26:02AM -0800, Erick Torres wrote:
> Thanks! Appreciate the clarification

Any documentation that needs clarification?