Rahix/avr-hal #650: Projects making floating-point operations fail building during linking after toolchain version bump

davidpolverari, 2025-04-17:
After the the update of Rust toolchain to nightly-2025-03-03 (https://github.com/Rahix/avr-hal/commit/9e96d0efe67367749b43084fcbe474649d6b62cf), projects that perform floating-point operations are failing, due to undefined references to GCC soft float library routines (eg: `__divsf3`, `__gesf2` and `__gtsf2`) during the linking process.

For instance, this line:

```rust
// duration is a u32
let distance = duration_us as f32 / 58.0;
```

Generates the following error:

```
src/main.rs:230: undefined reference to `__divsf3'
```

While these lines:

```rust
// dist is the same value as the `distance` above
tdd.write(dist as u8);
```

```
src/main.rs:280: undefined reference to `__gesf2'
src/main.rs:280: undefined reference to `__gtsf2'
```

Those errors didn't happen when using the previous Rust nightly version, along with using the JSON spec files.

Rahix, 2025-04-17:
Cc: @Patryk27

Patryk27, 2025-04-17:
Could you run `avr-gcc --version` and show the results?

Patryk27, 2025-04-17:
Ah, also, could you check on newer nightly?

davidpolverari, 2025-04-18:
> Could you run `avr-gcc --version` and show the results?

```console
$ avr-gcc --version
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

davidpolverari, 2025-04-18:
> Ah, also, could you check on newer nightly? (e.g. from yesterday or so)

Thanks a lot! I worked when using `nightly-2025-04-17`! I will leave it to you closing this or not, as I'm not sure whether it should remain open to track it until you update the nightly version on the project.

By the way, thank all you guys (@Rahix, @Patryk27, et al) by the work on `avr-hal` and other crates in the Rust embedded ecosystem! You all rock!

Patryk27, 2025-04-18:
> Thanks a lot! I worked when using nightly-2025-04-17!

Cool, good to know!

fwiw, bumping your avr-gcc might've helped as well, IIRC avr-gcc 5.4 doesn't provide floating-point intrinsics for AVR (I've got avr-gcc 14.2.1, for example) -- but with newer rustc this doesn't matter, since we supply the required stuff out-of-the-box 🙂

Rahix, 2025-04-18:
@Patryk27 should we upgrade everything to `nightly-2025-04-17` then?

Patryk27, 2025-04-18:
There are no known regressions, so I'd say so!

Rahix, 2025-04-18:
Created PR #651 :)

closed 2025-04-28 by Rahix