Rahix/avr-hal #660: Switch pin beetween input and output mode

zaqxsw-dev, 2025-05-12:
How to make it work? 

```rust
#![no_std]
#![no_main]

use core::cell::RefCell;

use arduino_hal::{
	delay_ms, delay_us,
	hal::port::{Dynamic, PD2},
	port::{
		mode::{Floating, Input, PullUp},
		Pin,
	},
};
use avr_device::interrupt;
use embedded_hal::digital::InputPin;
use panic_halt as _;

type Console = arduino_hal::hal::usart::Usart0<arduino_hal::DefaultClock>;
static CONSOLE: interrupt::Mutex<RefCell<Option<Console>>> =
	interrupt::Mutex::new(RefCell::new(None));

macro_rules! println {
    ($($t:tt)*) => {
        interrupt::free(
            |cs| {
                if let Some(console) = CONSOLE.borrow(cs).borrow_mut().as_mut() {
                    let _ = ufmt::uwriteln!(console, $($t)*);
                }
            },
        )
    };
}

fn put_console(console: Console) {
	interrupt::free(|cs| {
		*CONSOLE.borrow(cs).borrow_mut() = Some(console);
	})
}

#[arduino_hal::entry]
fn main() -> ! {
	let dp = arduino_hal::Peripherals::take().unwrap();
	let pins = arduino_hal::pins!(dp);
	let serial = arduino_hal::default_serial!(dp, pins, 57600);
	put_console(serial);

	let mut led = pins.d13.into_output();
	led.set_high();

	let mut dht_pin = pins.d2.into_opendrain_high();

	loop {
		println!("qqqqqqq");
		led.toggle();

		dht_pin.set_low();
		delay_ms(18);
		dht_pin.set_high();
		delay_us(40);
		dht_pin = dht_pin.into_pull_up_input();

		match read_dht11(&mut dht_pin) {
			Ok((humidity, temperature)) => {
				println!("qweqwe");
				println!(
					"Humidity: {}%, Temperature: {}°C",
					humidity as u32, temperature as u32
				);
			}
			Err(e) => {
				println!("Error reading DHT11: {}", e);
			}
		}

		arduino_hal::delay_ms(1000);
	}
}

fn read_dht11(dht_pin: &mut Pin<Input<PullUp>, PD2>) -> Result<(f32, f32), &'static str> {
	let mut data = [0u8; 5];

	println!("qqqqqqq2");
	// Wait for response signal
	while dht_pin.is_high().unwrap_or(false) {}
	println!("qqqqqqq25");
	while dht_pin.is_low().unwrap_or(false) {}

	println!("qqqqqqq3");
	// Read 40 bits of data
	for i in 0..40 {
		while dht_pin.is_low().unwrap_or(false) {}
		let mut length = 0;
		while dht_pin.is_high().unwrap_or(false) {
			length += 1;
			delay_us(1);
		}

		data[i / 8] <<= 1;
		if length > 30 {
			data[i / 8] |= 1;
		}
	}
	println!("qqqqqqq4");

	// Validate checksum
	if data[4] != (data[0] + data[1] + data[2] + data[3]) {
		return Err("Checksum validation failed");
	}

	let humidity = data[0] as f32 + (data[1] as f32 / 10.0);
	let temperature = data[2] as f32 + (data[3] as f32 / 10.0);

	Ok((humidity, temperature))
}

Rahix, 2025-05-13:
Please describe what problem you are hitting and which piece of your code you think is responsible.  Don't expect others to just decipher your code to understand what's your problem.

zaqxsw-dev, 2025-05-13:
Excuse me. Mistake is here

`error[E0382]: borrow of moved value: `dht_pin`
   --> src/main.rs:99:3
    |
93  |     let mut dht_pin = pins.d2.into_opendrain_high();
    |         ----------- move occurs because `dht_pin` has type `avr_hal_generic::port::Pin<OpenDrain, PD2>`, which does not implement the `Copy` trait
94  |
95  |     loop {
    |     ---- inside of this loop
...
99  |         dht_pin.set_low();
    |         ^^^^^^^ value borrowed here after move
...
103 |         let mut dht_pin2 = dht_pin.into_pull_up_input();
    |                                    -------------------- `dht_pin` moved due to this method call, in previous iteration of loop`

If in short, i misunderstood how to correct borrow pins value if i need switch pin mode in loop.

Rahix, 2025-05-13:
So in your particular situation, you can solve this by explicitly converting the pin back into its original mode:

```rust
let mut dht_pin_out = pins.d2.into_opendrain_high();
loop {
    dht_pin_out.set_low();

    let mut dht_pin_in = dht_pin_out.into_pull_up_input();

    // then, later
    dht_pin_out = dht_pin_in.into_opendrain_high();
}
```

For more dynamic situations or when you only hold a reference to a pin, there will be #205 which unfortunately is not finished yet.