Rahix/avr-hal #661: How are interrupt  handlers implemented by default?

n-eq, 2025-05-12:
Hello,

I'm trying to understand how things work when programming AVR microcontrollers in Rust, and I'm currently struggling find out where the "default" implementation of the MCUs' interrupt handlers are.

For example, when using the ATmega328p, one can define this function:
```rust
#[avr_device::interrupt(atmega328p)]
unsafe fn TIMER0_OVF() {
    // ...
}
```
as the interrupt handler of TIMER0 OVF.

My question is: what if this function isn't defined by the user?
After inspecting the generated assembly when no custom handler is defined, I can see the handlers addresses contain jumps to 0x0 (reset_handler).
Is this behaviour somehow enforced by the crate? If so, where is this done?

Thanks for your help!

Rahix, 2025-05-13:
Why did you close this issue again?  If you managed to find out yourself, it would be nice to at least document the answer for everyone else as well.  If the question still remains, let me know.

n-eq, 2025-05-13:
Hi Rahix, as it turned out the question wasn't related to the crate, and I thought the answer was obvious.
When undefined, interrupt handlers are set to jump to 0x0 by default (that is the reset handler)