Rahix/avr-hal #664: Missing data when reading from USART on Arduino Uno

katediadechko, 2025-05-25:
I'm trying to buffer USART data before processing them on Arduino Uno. The modified test can be found [here](https://github.com/katediadechko/avr-hal/blob/uno-usart-test/examples/arduino-uno/src/bin/uno-usart.rs).
The test reads the input in 10-byte chunks (or until EOL) and prints the chunks back.
Somehow, the data gets missing after the first chunk. For instance, here are two inputs and the corresponding results.

```
Hello from Arduino!
01234567890123456789
Got 0123456789
Got 01
abcdefghijklmnopqrstuvwxyz
Got abcdefghij
Got klxyz
Got
```

I translated the test into an [Arduino sketch](https://gist.github.com/katediadechko/1f8766a955ea1d9e9498f50180c4f7d0), and it seems to work just as expected - the device prints 10-byte chunks plus `\r`, `\n`.

```
Hello from Arduino!
Got 0123456789
Got 0123456789
Got 
Got 
Got abcdefghij
Got klmnopqrst
Got uvwxyz
Got 
```

I can imagine that my Rust code is not correct. Will highly appreciate any help.

bdew, 2025-06-08:
I'm trying to translate some C++ arduino code to rust and running into the same thing.

Is there an example somewhere on how to do this correctly? With buffering incoming data while it's being processed or data is written back?

i see there's `listen` that says it's enables an interrupt but i don't see how to actually receive the interrupt?

bdew, 2025-06-08:
So it's not something provided by the hal at all? 

That makes the whole serial interface not very useful if i have to deal with low level register stuff just to be able to receive more than a singly byte